adafruit/Adafruit_BME680 #1: The example can not be compiled for an ESP32

micropet, 2017-11-02:
I am using an ESP32 Dev board. The BME680 is connected via I2C.

D:\arduino-1.9\libraries\Adafruit_BME680\Adafruit_BME680.cpp: In function 'int8_t spi_read(uint8_t, uint8_t, uint8_t*, uint16_t)':

D:\arduino-1.9\libraries\Adafruit_BME680\Adafruit_BME680.cpp:348:88: error: 'int8_t spi_read(uint8_t, uint8_t, uint8_t*, uint16_t)' was declared 'extern' and later 'static' [-fpermissive]

 static int8_t spi_read(uint8_t cspin, uint8_t reg_addr, uint8_t *reg_data, uint16_t len) {

                                                                                        ^

In file included from D:\arduino-1.9\libraries\Adafruit_BME680\Adafruit_BME680.cpp:18:0:

D:\arduino-1.9\libraries\Adafruit_BME680\Adafruit_BME680.h:34:8: note: previous declaration of 'int8_t spi_read(uint8_t, uint8_t, uint8_t*, uint16_t)'

 int8_t spi_read(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len);

        ^

D:\arduino-1.9\libraries\Adafruit_BME680\Adafruit_BME680.cpp: In function 'int8_t spi_write(uint8_t, uint8_t, uint8_t*, uint16_t)':

D:\arduino-1.9\libraries\Adafruit_BME680\Adafruit_BME680.cpp:387:89: error: 'int8_t spi_write(uint8_t, uint8_t, uint8_t*, uint16_t)' was declared 'extern' and later 'static' [-fpermissive]

 static int8_t spi_write(uint8_t cspin, uint8_t reg_addr, uint8_t *reg_data, uint16_t len) {

                                                                                         ^

In file included from D:\arduino-1.9\libraries\Adafruit_BME680\Adafruit_BME680.cpp:18:0:

D:\arduino-1.9\libraries\Adafruit_BME680\Adafruit_BME680.h:35:8: note: previous declaration of 'int8_t spi_write(uint8_t, uint8_t, uint8_t*, uint16_t)'

 int8_t spi_write(uint8_t dev_id, uint8_t reg_addr, uint8_t *reg_data, uint16_t len);

        ^

D:\arduino-1.9\libraries\Adafruit_BME680\Adafruit_BME680.cpp: In function 'uint8_t spi_transfer(uint8_t)':

D:\arduino-1.9\libraries\Adafruit_BME680\Adafruit_BME680.cpp:421:38: error: 'uint8_t spi_transfer(uint8_t)' was declared 'extern' and later 'static' [-fpermissive]

 static uint8_t spi_transfer(uint8_t x) {

                                      ^

In file included from D:\arduino-1.9\libraries\Adafruit_BME680\Adafruit_BME680.cpp:18:0:

D:\arduino-1.9\libraries\Adafruit_BME680\Adafruit_BME680.h:36:9: note: previous declaration of 'uint8_t spi_transfer(uint8_t)'

 uint8_t spi_transfer(uint8_t x);

         ^

exit status 1
Fehler beim Kompilieren für das Board ESP32 Dev Module.

ladyada, 2017-11-02:
hiya try the latest commit
thx!

closed 2017-11-02 by ladyada

micropet, 2017-11-02:
Fine. it works now.
Many Thanks.