adafruit/Adafruit_BME680 #3: Feedback: Test on custom board.

hecko, 2017-11-06:
I can confirm this code working with BME680 bought from Mouser on a custom board with ATMEGA328P without external crystal using I2C. Arduino IDE 1.8.3

G6EJD, 2017-11-08:
The 0-500 range is the output of the Bosch IAQ algorithm (external library) not the so-called gas resistance which I notice Adafruit has converted to Kohms now with a divide /1000. My sensor typically outputs in the range 50000-5000000 or 50K-500K. You can make your own IAQ, I used 0-25% humidity plus 0-75% gas then scaled that to 0-500. There is no IAQ standard afaik.

ladyada, 2017-11-08:
that is correct, the IAQ algorithm they use is under lock & key, and only available in precompiled library formatl. @G6EJD where'd you get the idea for the hum+gas mix? i found some readings i put here https://learn.adafruit.com/adafruit-bme680-humidity-temperature-barometic-pressure-voc-gas/downloads which may be useful

G6EJD, 2017-11-09:
Many organisations use the 0-500 IAQ index, but no one says how it’s comprised, so I figured an environment is affected by humidity (very low is not good, 40% is perfect or very high is not good) and then for gas a nominal reading with no obvious gas around is good, I chose the range 50K is very bad and 500K very good combined and scaled the results to 100% then as an add-on scaled it to the de-facto IAQ. The code is in github G6EJD.

janzieg, 2017-12-07:
These guys managed the integrationof the Bosch BSEC library that does all the calculations: https://github.com/DFRobot/DFRobot_BME680.
Would be nice to have this within here.

(Main point: BSEC is precompiled and platform specific).

ESP32DE, 2018-02-01:
hi @ladyada 
thanks for the lib. 
i checked the BOSCH Driver SRC and [did a small PR ](https://github.com/BoschSensortec/BME680_driver/pull/16) on https://github.com/BoschSensortec/BME680_driver

i [did try your lib today](https://twitter.com/eMbeddedHome/status/959146544388820993) on [Olimex ESP32Pro Board](https://esp32.com/viewtopic.php?f=17&t=2227#p10519) and my output looks like this:
[image]

it work's

now comes fine tune and not sure, for gas i will use a referenz table
this values then must be a referenz that you have to measuring by "live" i think.
but i will read first datasheet to the end :) 

best wishes
rudi ;-)

dherrada, 2020-06-24:
Closing since this seems more or less resolved

closed 2020-06-24 by dherrada