adafruit/Adafruit_BME680 #7: Hidden Datatypes TVOC and eCO2

jttaylor86, 2018-03-26:
Hello,

would it be possible to create a library where the hidden datatypes, TVOC and eCO2 are easily readout in a arduino sketch?

Thanks

ladyada, 2018-03-26:
nope, they're not hidden, they're unimplemented!
see https://github.com/adafruit/Adafruit_BME680/issues/3 
:(

closed 2018-03-26 by ladyada

jttaylor86, 2018-03-27:
Hello there,

they are implemented however I am not smart enough to change the code. Have
a look at hidden_datatypes.h.

Here is a link to an internal release.
https://drive.google.com/drive/folders/1CM039lG9IcAkEAeK6HVUIJseVdomZWKg?usp=sharing

I would be very grateful if you could magic up an Arduino library with TVOC
and CO2 readings as I can't.

Thanks,

Justin

ladyada, 2018-03-27:
the BSEC/API is under NDA, if you look at the license, we cannot publish/distribute it. so we cannot add that to this library.