adafruit/Adafruit_BME680 #15: Too long a delay to wait for a reading?

tlongeri, 2018-08-23:
I am talking about what can be seen on lines 327-334 of Adafruit_BME680.cpp (`endReading` function):

```cpp
  unsigned long now = millis();
  if (meas_end > now) {
    unsigned long meas_period = meas_end - now;
#ifdef BME680_DEBUG
    Serial.print("Waiting (ms) "); Serial.println(meas_period);
#endif
    delay(meas_period * 2); /* Delay till the measurement is ready */
  }
```
Specifically line 333, `delay(meas_period * 2);`, which has a mysterious factor of 2 which seems completely random and magic, and leaves me scratching my head.

I convinced myself there wasn't some hardware reason for that because it also gives it inconsistent behavior. If I run:

```cpp
bme.setGasHeater(320,5000); // 320 ºC for 5000 ms
bme.beginReading();
bme.endReading();
```

Then I will be getting my reading after just over 10 seconds (`endReading` would delay for about 5000 * 2 = 10000 ms). However, if I run the following:

```cpp
bme.setGasHeater(320,5000); // 320 ºC for 5000 ms
bme.beginReading();
delay(2000);
bme.endReading();
```

Then I get my reading after just over 8 seconds (`endReading` would delay for about 3000 * 2 = 6000 ms, plus the initial 2000 ms delay) instead of the previous 10 seconds.

In both of the above cases I would have expected to have my reading done after 5 seconds (and certainly not to have them take different times).

ladyada, 2018-09-23:
i think we just doubled it for safety ... see the async example, maybe that would help you if you want to minimize time spent waiting?

caternuson, 2022-11-21:
Closing due to lack of response. Try the suggested async approach.

closed 2022-11-21 by caternuson