adafruit/Adafruit_BME680 #17: cannot turn off gas sensor once turned on

pabloandresm, 2018-10-24:
Hello,

If I start a project with ".setGasHeater(0, 0);" then the gas sensor returns "0" as value, which is normal because it was setted off.

If at one point I set the sensor on (with .setGasHeater(320,150) for example), it correctly goes on.

But after that it cannot be set to off anymore. The .setGasHeater(0,0) issued later does not turn the gas sensor off.

I've tried to look at the code to find for a logical 'or' not being 'anded with the opposite ~' or any error like that but found nothing so far.

What do you think?
Thanks

pabloandresm, 2018-10-25:
I think I've found one bug.
At the end of the function ::beginReading() in file Adafruit_BME680.cpp, in the last "if" there is a missing "else gas_resistance = NAN;"

The correct lines would be:
  /* Avoid using measurements from an unstable heating setup */
  if (_gasEnabled) {
    if (data.status & BME680_HEAT_STAB_MSK) {
      //Serial.print("Gas resistance: "); Serial.println(data.gas_resistance);
      gas_resistance = data.gas_resistance;
    } else {
      gas_resistance = 0;
      //Serial.println("Gas reading unstable!");
    }
  }  else gas_resistance = NAN;   // <<====== MISSING LINE

So when I turn off the gas sensor in the middle of a project the reading will return zero.

But I think there is still a bug, because after turning the gas sensor off, the temperature reading does not go down as it should (because the gas heater is not turned on anymore).
So there is still an error somewhere.

caternuson, 2018-10-28:
The Arduino library doesn't have explicit heater on/off control. Calling `.setGasHeater(0, 0)` only disables gas measurement:
https://github.com/adafruit/Adafruit_BME680/blob/master/Adafruit_BME680.cpp#L397
The underlying Bosch library does though:
https://github.com/adafruit/Adafruit_BME680/blob/master/bme680.c#L449

I'm not sure I follow your second post. Are you wanting a return of NAN instead of 0 when the sensor is disabled?

caternuson, 2018-10-29:
Are you talking about this `if` block?
https://github.com/adafruit/Adafruit_BME680/blob/master/Adafruit_BME680.cpp#L370

If not, can you link to it directly by line.

jib218, 2019-01-05:
Any plans of adding this feature?

ladyada, 2019-01-05:
we won't be getting to it any time soon - if you'd like to add it and submit a PR - we'd check it out :)

jib218, 2019-01-06:
HTH: #19

:-)

ladyada, 2019-01-06:
thanks u!

closed 2019-01-06 by ladyada