adafruit/Adafruit_BME680 #30: problem with time critical application

silvan2468, 2019-04-19:
Dear all

Thanks a lot for your BME680-library. It works almost fine for me on a particle photon. There is just one problem on my application. I do a lot of controlling stuff with 500 Hz sampling frequency. It is a self balancing inverted pendulum.
So everytime I perform a sensor reading, the controll algorythm crashes for about a half second. I found that if I, just for a test, uncomment the delay in line 337 in the class "Adafruit_BME680" it is better, but anyway disturbing for about 100ms my control algorythm.
Do you know a solution how your code is also usable in a time critical application?

Thanks and kind regards
Silvan

siddacious, 2019-04-19:
@silvan2468 Can you give more details about what you mean by "the control algorithm crashes for about half a second"? Is your code crashing or the BME680 driver? Usually when things crash they stay crashed.

adrionics, 2019-04-19:
Silvan, are you using it based on the "async" sample? You need to check that the reading is ready by calling beginReading() and see that it is less than millis() before calling endReading(). If you call endReading() before it is ready, it will use a delay() and block till the sensor is ready with a reading (potentially causing your problem).

silvan2468, 2019-04-19:
Dear all

Thanks a lot for the fast answers.

1. siddacious: crashing is the wrong word. You're right. The control algorythm seams stoped for a half second. This is acting as uncontrolled movement and falling of the inverted pendulum.

2. thanks adrionics for the idea with the async example. I tried it with this and this improved the problem so far, that the pendulum is not falling completely, but is uncontrolled pendeling for about 100ms.
The serial reading shows:

"Reading startet at 180652 and will finish at 180833
You can do other work during BME680 measurement.
Reading completed at 180656.
Temperature 29,09 *C
...."

So I think the reading is done all right. If the millis are right, the measureing time was just 4 ms. This would be two control loops (500 Hz = 2ms). I think it is possible that these missing control loops have the described effect of uncontrolled pendeling.
I call the example exactly as it is in the example every 10s, just uncomented the delay(50).

So if it is possible to eliminate this last small interrupt of my control loop, this would be realy nice.
I read normaly (not for this test) also a lot of other sensors (INA219, APDS9960,...) through the i2c, but no one makes me such problems.
Are there perhaps some while-loops in these 4ms? Breaking out my control loop?

Thanks a lot and kind regards
Silvan

silvan2468, 2019-04-20:
I've got the solution and the problem.
The problem was that I executed the reading in a TASK of 10 seconds. When i changed it to the loop, there was no more conflict. I'm not sure why, but is it perhaps, that the task force the execution while the loop does only as fast as possible?

siddacious, 2019-04-20:
Glad you found a solution! Your project sounds very interesting, you should consider showing it on Show and Tell.

closed 2019-04-20 by siddacious

adrionics, 2019-04-21:
@silvan2468 , I am looking into some other potential issues with the driver. It uses delay even with the async example. You could add some Serial.println to the driver just before it calls delay to see where it may still call delay when using "async" method. That may help you to understand more where it is doing this.

adrionics, 2019-04-22:
In the function `int Adafruit_BME680::remainingReadingMillis(void)`

Instead of `int remaing_time = (millis() - _meas_start) - (int)_meas_period;`

It should be  `int remaing_time = (_meas_start + (int)_meas_period) - millis();`

silvan2468, 2019-04-22:
Dear adrionics

Thanks for further help. 
But for me it works now fine.

Remaining time: this seams to be the same warriorofwire has done in his open pull-request. I have already done it, but didn't realy recognised a huge improvement. 

Kind regards
Silvan

adrionics, 2019-04-22:
You will not notice an improvement, but it is causing the sensor to wait more using delay. Lets say you get 182ms back from calling beginReading, after that time, you call endReading, then it will wait another 182ms using delay for no reason.