adafruit/Adafruit_BME680 #34: Temperatur readings to high

chmasi, 2019-09-05:
A wonderfull good morning

Doing some research in zhe www I found this for the BME680

_

> That's normal. The sensor is not factory calibrate for temperature. You need to add a temperature offset.
> Not sure if the library supports.
> 
> Looking at a reference thermometer, we can see that our temperature is indeed a few degrees to high. Does that mean the temperature sensor inside the BME680 is inaccurate?
> Actually no, it very accurately measures the temperature exactly where it is located on the board. But there also is the issue: our board as most devices contains some heat sources (e.g., MCU, WiFi chip, display, …). This means the temperature on our board is actually higher than the ambient temperature. Since the absolute amount of water in the air is approximately constant, this also causes the relative humidity to be lower on our board than elsewhere in the room.
> As BSEC cannot know in which kind of device the sensor is integrated, we have provide some information to the algorithm to enable it to compensate this offset. In the simplest case, we have to deal with an embedded device with a constant workload and approximately constant self-heating. In such a case, we can simply supply a temperature offset to BSEC which will be subtracted from the temperature and will be used to correct the humidity reading as well.
> 
> from the Integration Guide - Bosch Software Environmental Cluster (BSEC)
_

Is this already integrated into the Adafruit Library, if not is there an easy way how to do the compensation? If yes how can I set the correction?

Thanks
Martin

ladyada, 2019-09-22:
there's no built in offset or calibration - we read directly from the board, you would add any correction to your arduino sketch

sblantipodi, 2019-11-06:
@ladyada  sorry but I must admit that those sensor are crap.
paid 90€ for three non calibrated sensors.

who need a non calibrated sensor?
your price premium does not worth any money over the chinese counterpart.

hecko, 2019-11-06:
@sblantipodi I must back @ladyada up here. 

the sensors ARE calibrated but they are not calibrated to use in your specific application - meaning to properly use the sensors you must know how and where to mount them, how to compensate the offset for specific use case. this is not something Adafruit can do - this is something that you, asn an engineer, must do. Metrology is a very complex field and you need and must understand at least some basics to understand how complicated it is. We sell calibrated measurement temperature sensors for hundreds of eur and this is not because of the cosst of the sensor - this is because to calibrate the thermometer is very costly (lots of additional equipment) and time consuming process.

sblantipodi, 2019-11-06:
@hecko my specific application is the general purpose application and the application where a good sensor should be calibrated on.

FREE AIR.

we don't epect stellar accuracy but neither 3/4°C errors.

hecko, 2019-11-06:
@sblantipodi free air is probably the most difficult to measure from all the elements!

Just look at some of the radiation shields and their costs from companies like Davis or Barani designs - the plastic bit around the sensor costs hundreds of euros because they know how to make the air inside the radiation shield static enough and sensor isolated from radiation to make a proper reading.

sblantipodi, 2019-11-06:
@hecko if it is so difficult, why don't advertise people that those 30€ sensors are not reliable for ambient temperature readings and no way good to be used as a thermostat?

the strange things is that in my life I had a lot of "non expensive thermostat" and no one had all those overheating problems.

hecko, 2019-11-06:
> @hecko if it is so difficult, why don't advertise people that those 30€ sensors are not reliable for ambient temperature readings and no way good to be used as a thermostat?

Because this is a development board with a sensor and the development board is not calibrated. The sensor is. Measurement devices dont work the way you think they do. The non expensive sensors you find elsewhere might give you possibly more precise reading in 20-30 degree range (C) but will drift greatly outside of this range. And why? Because they are calibrated to be used in this very specific range! How do you even know the sensor does not read the value correctly?! What is your reference thermometer?

@sblantipodi get a reference thermometer and make a 3 or 4 point calibration curve for your application - its theoretically very easy so you can do it is a jiffy, just mind that that everything inside you device affects the measurement of the temperature - the DCDC's, the substrate you use on the pcb, the plastics you use, the radiation from objects around, even the AC cables and temperature characteristics of the walls your sensor is mounted on. I can see that you are building an arduino-based thermostat - if this is your use case make sure you place the sensor somewhere where its isolated from the noise of electronics and AC cables running into the device. 

Get out of your comfort zone and implement PT100 senzor reading device. Its HARD to do well!

This process is called engineering. As harsh as this may seem - be a proper engineer and do not rant on a forum that the sensor is a crap! SIt down, read through the datasheet and make the compensation curve; then come back and rightfully brag about the engineering you have done!

sblantipodi, 2019-11-06:
@hecko don't understand what is the added value of the adafruit sensors over the chinese one.
and don't understand why a dev board should not e factory calibrated.

I understand that dev board are used by devs who can correct the readings with another sensor but why don't factory calibrate the sensor with your specific breakout in free air?

still feel that I wasted my money with your products since I see no added value over the chinese product.

and ok, I can understand that there are a lot of variables but those variables can't create a 4°C error like on your breakout.

hecko, 2019-11-06:
@sblantipodi I have no affiliation with Adafruit

Im going to take some additional time to answer your question.

>@hecko don't understand what is the added value of the adafruit sensors over the chinese one.

Lots! One of them is that someone like me is spending his time to answer your questions on this public Adafruit forum. Because there is one. ANd you have several hundred pages of datasheet to go with the sensor. Also the sensor is from a certified distributor I would assumed who has affiliation with bosch, who is the manufacturer.

>and don't understand why a dev board should not e factory calibrated.

Because they would cost much, much more and development boards are not meant to be calibrated because the calibration would not make sense.

>I understand that dev board are used by devs who can correct the readings with another sensor but why don't factory calibrate the sensor with your specific breakout in free air?

No, dev boards are used for developers to evaluate the product - like you are doing right now. Engineers need to know how hard it is to integrate this sensor into their future designs. To measure the product, to evaluate whether it is worth extra effort to integrate this or other sensor into their products. To write firmware. Etc.

>still feel that I wasted my money with your products since I see no added value over the chinese product.

Ok, if you feel that way then you have several possibilities to pursue now including returning the product (as by law). Just be aware that there is no way around this. If you ever want to make your thermostat or any other temperature measuring product a product you WILL HAVE TO go through the calibration process anyways. There are several ISO standards you have to comply with when measuring anything of this sort! Without them product may be illegal to sell.

>and ok, I can understand that there are a lot of variables but those variables can't create a 4°C error like on your breakout.

I assume its because the BOSCH sensor is more sensitive to radiation and external factors than other sensors. 
Read this and evaluate: https://www.mouser.sk/datasheet/2/783/BST-BME680-DS001-1509608.pdf its only 60 pages long and critical to evaluating in your application I assume.

sblantipodi, 2019-11-06:
me, like most of adafruit users does not have a reference thermometer.

what is the "real accuracy" that we can expect from those adafruit sensors?
I mean, the real difference between the real 0°C ambient temp and the sensors readings?

I don't care about the accuracy if I have an uncalibrated sensor.
What is the maximum deviation between the real temp and the sensor reading?

hecko, 2019-11-06:
Read page 12 of the datasheet. +-0.5C, +-1C respectively.

sblantipodi, 2020-06-16:
is it normal that
`boschBME680.setIIRFilterSize(BME680_FILTER_SIZE_3); `
causes a temperature increase of at least 1°C over
`boschBME680.setIIRFilterSize(BME680_FILTER_SIZE_0); `
?

thanks

caternuson, 2021-07-22:
Closing. Not a library issue.

closed 2021-07-22 by caternuson