adafruit/Adafruit_BME680 #36: I2C pins

oxedgar, 2019-10-19:
How can i sellect specific pins for I2C bus to use this library sensor with non standard pins?

ladyada, 2019-10-19:
https://github.com/adafruit/Adafruit_BME680/blob/master/Adafruit_BME680.h#L48

closed 2019-10-19 by ladyada

N0ury, 2023-02-23:
I would like to use a BME680 in I2C mode but with other pins than A4 and A5.
I use an esp8266-12f. It has many GPIO pins that can be used for I2C communication.
I have googled a lot, and all posts give the same example that uses the default pins.
I have found how to change the address, but nothing about I2C pins.
I know that the Wire lib permits this (with the begin command), but I really don't know how to use it with the code given as examples.
I have found this "issue", but I don't understand the answer. What in line 48 should gives me an answer?
A little more informations are necessary. I didn't want to open a new issue, I'm I wrong?
ps: for some reason, I can't use the default pins.