adafruit/Adafruit_BME680 #37: Gas resistance reading is heavily dependent on polling interval

bobnecat, 2020-04-16:
As the header states, there is an issue with logic of gas reading part of the code "gas_resistance".

The output of this function is heavily dependent on frequency of polling the sensor. For example if I call this function every 3 seconds, the output would be in the range of 100kOhms, however dropping down polling interval to 1 minute, the average output drops to ~20kOhm while performing reading every 10 minutes result in inconsistent output between 1 and 5kOhms. 

I've been reading through the library code, however unable to identify what causes it. Most likely there is a bug in the way heating (temperature and duration) of gas sensor plate is realized in this library.

Manuel-Sensate, 2020-04-23:
I just stumbled over this issue as well.. did you find an solution?

bobnecat, 2020-04-27:
Nope, moreover I think this sensor or the library implementation do not make any sense without original BSEC library. I've switched my polling interval to every 3 seconds (to mimic polling in BSEC library) and my results in the long term are still all over the place. The sensor definitely reacts to changes in conditions, however those reactions are not consistent and don't make much sense overall.

xjzp, 2020-04-29:
@bobnecat - I had just given up here after seeing that behavior as well, at least I know I'm not alone on that.  I thought I had damaged mine running it perhaps too much.

CptHolzschnauz, 2021-09-21:
I can confirm. It changes also with ages. A new (but burnt in for 24h) gives a tenth of an old sensor. As long as Bosch is playing games with secret algorithms it's very difficult to read out a sensible value.

bobnecat, 2026-02-14:
This was an old topic, but by now I have over 10 years worth of empirical data and experience for a good number of Bosch sensors (particularly - BME280, BMP and BME680) and the conclusion I came to is that they are just crap for the most part... 

BSEC external library requirement just does not make any sense, for one - they are already using digital logic IC inside of every BME module, they could have easily designed and implemented algorithms processing internally without the necessity of loading external MCU with nonsense operations. Second  - even if you follow their guidelines and use BSEC and other bs they require you to take care of, you still get AQI data that resembles going to local psychic & palm reader to understand what you did wrong in life that it entangled you with this sensor - the data output although does show some changes and shifts, you just have to try and guess what it reacts to and what all that means...  
 
As for BME/BMP280 - while 10+ years ago having a BMP280 replace primitive and unreliable DHT11/22 was such an excitement, in today's variety of sensors there are just so many more reliable and accurate options.