adafruit/Adafruit_BME680 #40: Latest 1.0.8 and 1.1 broke BME680 readings.

sblantipodi, 2020-06-16:
As title.
this simple code works perfectly with lib version 1.0.7
```
      if (!boschBME680.performReading()) {
        Serial.println("Failed to perform reading :(");
        delay(500);
        return;
      }
      BME680["Temperature"] = serialized(String(boschBME680.temperature + tempSensorOffset,1));
      BME680["Humidity"] = serialized(String(boschBME680.humidity,1)); 
      BME680["Pressure"] = serialized(String(boschBME680.pressure/100,1)); 
```

it does not work on 1.1.0

can you fix it please?

sblantipodi, 2020-06-16:
@hoffmannjan I think that latest commits are yours :)

sblantipodi, 2020-06-19:
am I the only one with this problem?

dherrada, 2020-06-22:
Hey @sblantipodi Do you think you could let me know what hardware you're using and what (if anything) you're seeing in the serial terminal

sblantipodi, 2020-06-22:
@dherrada thanks for the answer, I appreciate it.
unfortunantly the microcontroller is running inside my wall and it's pretty difficult to debug the serial.

I'm running the genuine BME680 from adafruit on a Lolin D1 Mini ESP8266EX board.

dherrada, 2020-06-22:
Ah, well I don't happen to have an esp8266EX, so if you could find out what's actually causing the failure by connecting to a serial terminal, then we can take a look at it. I'd be glad to look at it, but something just not working isn't enough to go off to reproduce something. Do you have another esp you might be able to hook up the BME680 to?

sblantipodi, 2020-06-23:
1.0.7 works well, 1.1.0 no.
hope that Adafruit will chime in since they are the owner of the lib, of the break out board and they have ESP8266 boards as well :)

dherrada, 2020-06-23:
> 1.0.7 works well, 1.1.0 no.

If you want help, you need to let us know what the error message you're getting is.

> hope that Adafruit will chime in

I'm an intern working for Adafruit.

dherrada, 2020-06-23:
I'm assuming this is a duplicate of #41. If you can provide me with the error message and it turns out to be different from the error message in 41, I'll reopen this.

sblantipodi, 2020-06-24:
@dherrada 
I attached the serial to my pc, no error at all, code works well without issues but all readings from the sensors return 0

can you reopen the issue please? and fix it if you can xD

dherrada, 2020-06-24:
Well, that's odd. And you're certain everything's hooked up correctly?

sblantipodi, 2020-06-24:
> Well, that's odd. And you're certain everything's hooked up correctly?

some code, same everything, different lib version
1.0.7 works like a charm, 
1.1.0 read 0 on all sensors.

tried it multiple times now.

dherrada, 2020-06-24:
I think that since both you and the person who opened #41 had the issue hit after the same release, it could be the same issue presenting itself in different ways. I'll focus on 41 first, since we've got more debug info to work off of, and then if the fix we find for that doesn't work for this issue, we can look more in-depth into what's going on here.

sblantipodi, 2020-06-24:
same problem as in #41 

you changed constructors without noticing breaking changes.

in my case the begin() function changed from this
    Wire.begin();
to this:
    _wire->begin();

but no one initialized the _wire, the constructor is not initializing it.

please fix your code examples and you will fix the problem ;)

have you tried to do a single test of your "code examples" by your own on your board before the release?

siddacious, 2020-06-25:
> please fix your code examples and you will fix the problem ;)
> 
> have you tried to do a single test of your "code examples" by your own on your board before the release?
@sblantipodi Please keep your comments civil, respectful, and constructive

hecko, 2020-07-08:
I can confirm that as of today's masters branch of the library I can reliably get BME680 data from the sensor,although running on ATMEGA328P

dherrada, 2020-07-08:
Awesome! good to know

tannewt, 2020-07-08:
Can we close this then?

closed 2020-07-08 by ladyada