adafruit/Adafruit_BME680 #41: BME680 failure

Idahowalker, 2020-06-23:
I been running a BME 680 since Feb 11th 2020.

This AM the ESP32 running the 680 shows this error,
Backtrace: 0x4000c2e4:0x3ffe3b80 0x400d2331:0x3ffe3b90 0x400d10bb:0x3ffe3bb0 0x400d6eaf:0x3ffe3bd0 0x400820f9:0x3ffe3bf0 0x400822d5:0x3ffe3c20 0x4007906f:0x3ffe3c40 0x400790d5:0x3ffe3c70 0x400790e0:0x3ffe3ca0 0x400792a9:0x3ffe3cc0 0x400806ea:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Which translates to this error:
"0x400d2331: Adafruit_BME680::Adafruit_BME680(signed char, SPIClass*) at C:\Users\KESOURLA\Documents\Arduino\libraries\Adafruit_BME680_Library\Adafruit_BME680.cpp line 83
0x400d10bb: _GLOBAL__sub_I_BME_CS() at C:\Users\KESOURLA\Documents\Arduino\ESP32_BMP680_test/ESP32_BMP680_test.ino line 29
0x400d6eaf: do_global_ctors at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c line 451
0x400820f9: start_cpu0_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c line 367
0x400822d5: call_start_cpu0 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c line 238"

Here is my code

*
   Project, use solar cells to generate power
   2/2/2020

*/
#include <WiFi.h>
#include <WiFiClientSecure.h>
#include <PubSubClient.h>
#include "certs.h"
#include "sdkconfig.h"
#include "esp_system.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/timers.h"
#include "freertos/event_groups.h"
#include "esp_sleep.h"
//#include <SPI.h>
//#include <Adafruit_Sensor.h>
//#include "Adafruit_BME680.h"
#include "Zanshin_BME680.h"
//////
EventGroupHandle_t eg;
#define evtDoBME ( 1 << 2 ) //100
#define evtSetupBME_Complete ( 1 << 3 )
SemaphoreHandle_t sema_ReadBME680;
//////
WiFiClient wifiClient;
PubSubClient client(mqtt_server, 1883, wifiClient);
//////
Adafruit_BME680 bme( GPIO_NUM_5 ); // hardware SPI
///
////
void setup()
{

 //  eg = xEventGroupCreate();
//  SPI.begin();
//  sema_ReadBME680 = xSemaphoreCreateBinary();
//  xTaskCreatePinnedToCore( fDoBME, "fDoBME", 20000, NULL, 3, NULL, 1 ); // assigned to core
  //start this task last
//  xTaskCreatePinnedToCore( fDoTheThing, "fDoTheThing", 40000, NULL, 5, NULL, 1 ); // assigned to core
} // end setup()
////
//void connectToMQTT()
//{
//  client.connect(clientID, mqtt_username, mqtt_password);
//  vTaskDelay( 200 );
//}
////
//void connectToWiFi()
//{
//  WiFi.begin( SSID, PWD );
//  vTaskDelay( 750);
//  while ( WiFi.status() != WL_CONNECTED )
//  {
//    vTaskDelay( 800 );
//  }
//}
//////
//void fDoTheThing( void * pvParameters )
//{
//  xEventGroupWaitBits (eg, evtSetupBME_Complete, pdTRUE, pdTRUE, portMAX_DELAY ); //
//  connectToWiFi();
//  connectToMQTT();
//  while (1)
//  {
//    xEventGroupSetBits( eg, evtDoBME ); // trigger tasks
//    xSemaphoreTake( sema_ReadBME680, portMAX_DELAY ); // wait for task to be done
//    client.disconnect();
//    WiFi.disconnect(true);
//    WiFi.mode(WIFI_OFF);
//    esp_sleep_enable_timer_wakeup( 1000000 * (60 * 5) ); // 1 second times how many seconds wanted
//    esp_deep_sleep_start();
//  } //while(1)
//  vTaskDelete ( NULL );
//} // void fDoTheThing( void * pvParameters )
//////
//void fDoBME ( void *pvParameters )
//{
//  while (!bme.begin())
//  {
//    log_i("Could not find BME680 sensor!");
//    vTaskDelay( 10 );
//  }
//  bme.setTemperatureOversampling(BME680_OS_8X);
//  bme.setHumidityOversampling(BME680_OS_2X);
//  bme.setPressureOversampling(BME680_OS_4X);
//  bme.setIIRFilterSize(BME680_FILTER_SIZE_3);
//  bme.setGasHeater(320, 150); // 320*C for 150 ms
//  float Temperature = 0.0f;
//  float Pressure = 0.0f;
//  float Humidity = 0.0f;
//  float GasResistance = 0.0f;
//  xEventGroupSetBits( eg, evtSetupBME_Complete ); // trigger task to begin
//  for ( ;; )
//  {
//    xEventGroupWaitBits (eg, evtDoBME, pdTRUE, pdTRUE, portMAX_DELAY ); //
//    log_i( "Signal strength %d ", WiFi.RSSI() );
//    Temperature = (bme.readTemperature() * 1.8f) + 32.0f; // (Celsius x 1.8) + 32
//    Pressure = bme.readPressure() / 133.3223684f; //converts to mmHg
//    Humidity = bme.readHumidity();
//    GasResistance = bme.readGas();
//    // log_i( "Temperature %f C, Pressure %f hPa, Humidity %f, IAQ %f", Temperature, Pressure, Humidity, GasResistance );
//    client.publish( "Home/oTemperature", String(Temperature).c_str() );
//    vTaskDelay( 500 ); // gives the Raspberry Pi 4 time to receive the message and process
//    client.publish( "Home/oHumidity", String(Humidity).c_str() );
//    vTaskDelay( 500 ); // no delay and RPi is still processing previous message
//    client.publish( "Home/oGas_Resistance", String(GasResistance).c_str() );
//    vTaskDelay( 500 );
//    client.publish( "Home/oPressure", String(Pressure).c_str() );
//    vTaskDelay( 1000 );
//    xSemaphoreGive ( sema_ReadBME680 );
//  } // for loop
//  vTaskDelete ( NULL );
//} // void fDoBME ( void *pvParameters )
////
////
void loop() {}
////
The error is happening with this line of code: Adafruit_BME680 bme( GPIO_NUM_5 ); // hardware SPI

If I comment out that line of code then I do not get an error.

Idahowalker, 2020-06-23:
I have also tried to use:
SPIClass SPI1( HSPI );
Adafruit_BME680 bme( GPIO_NUM_5, SPI1 ); // hardware SPI, kept getting an error about SPI1 is an int and not a SPI object.

dherrada, 2020-06-23:
@Idahowalker Hey, thanks for opening this issue. What version of the library are you using?

dherrada, 2020-06-23:
1.1.0 could be broken, so do you think you could test it with 1.0.8 and let me know if you get the same issue? Thanks!

Idahowalker, 2020-06-23:
Switched to using version 1.0.8
I get:
> 
st:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c2e4  PS      : 0x00060330  A0      : 0x800d234c  A1      : 0x3ffe3b80  
A2      : 0x00000000  A3      : 0x3ffc0464  A4      : 0x00000019  A5      : 0x00000000  
A6      : 0x00000003  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffe3b50  
A10     : 0x3ffc0724  A11     : 0x3f401b68  A12     : 0x00000001  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4000c2e4:0x3ffe3b80 0x400d2349:0x3ffe3b90 0x400d10d2:0x3ffe3bb0 0x400d6ec7:0x3ffe3bd0 0x400820f9:0x3ffe3bf0 0x400822d5:0x3ffe3c20 0x4007906f:0x3ffe3c40 0x400790d5:0x3ffe3c70 0x400790e0:0x3ffe3ca0 0x400792a9:0x3ffe3cc0 0x400806ea:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20
> 
Which translates to 
> 0x400d2349: Adafruit_BME680::Adafruit_BME680(signed char, SPIClass*) at C:\Users\KESOURLA\Documents\Arduino\libraries\Adafruit_BME680_Library\Adafruit_BME680.cpp line 84
0x400d10d2: _GLOBAL__sub_I_BME_CS() at C:\Users\KESOURLA\Documents\Arduino\ESP32_BMP680_test/ESP32_BMP680_test.ino line 29
0x400d6ec7: do_global_ctors at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c line 456
0x400820f9: start_cpu0_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c line 367
0x400822d5: call_start_cpu0 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c line 238

Here is the code I used
`/***************************************************************************
  This is a library for the BME680 gas, humidity, temperature & pressure sensor

  Designed specifically to work with the Adafruit BME680 Breakout
  ----> http://www.adafruit.com/products/3660

  These sensors use I2C or SPI to communicate, 2 or 4 pins are required
  to interface.

  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing products
  from Adafruit!

  Written by Limor Fried & Kevin Townsend for Adafruit Industries.
  BSD license, all text above must be included in any redistribution
 ***************************************************************************/
#include <Wire.h>
#include <SPI.h>
#include <Adafruit_Sensor.h>
#include "Adafruit_BME680.h"
//#define BME_SCK 13
//#define BME_MISO 12
//#define BME_MOSI 11
int8_t BME_CS = 5;
SPIClass SPI1(HSPI);
#define SEALEVELPRESSURE_HPA (1013.25)

//Adafruit_BME680 bme; // I2C
Adafruit_BME680 bme( BME_CS ); // hardware SPI
//Adafruit_BME680 bme( GPIO_NUM_5, GPIO_NUM_23, GPIO_NUM_19,  GPIO_NUM_18 );

void setup() {
  Serial.begin(115200);
  while (!Serial);
  Serial.println(F("BME680 test"));

  if (!bme.begin()) {
    Serial.println("Could not find a valid BME680 sensor, check wiring!");
    while (1);
  }

  // Set up oversampling and filter initialization
  bme.setTemperatureOversampling(BME680_OS_8X);
  bme.setHumidityOversampling(BME680_OS_2X);
  bme.setPressureOversampling(BME680_OS_4X);
  bme.setIIRFilterSize(BME680_FILTER_SIZE_3);
  bme.setGasHeater(320, 150); // 320*C for 150 ms
}

void loop() {
  if (! bme.performReading()) {
    Serial.println("Failed to perform reading :(");
    return;
  }
  Serial.print("Temperature = ");
  Serial.print(bme.temperature);
  Serial.println(" *C");

  Serial.print("Pressure = ");
  Serial.print(bme.pressure / 100.0);
  Serial.println(" hPa");

  Serial.print("Humidity = ");
  Serial.print(bme.humidity);
  Serial.println(" %");

  Serial.print("Gas = ");
  Serial.print(bme.gas_resistance / 1000.0);
  Serial.println(" KOhms");

  Serial.print("Approx. Altitude = ");
  Serial.print(bme.readAltitude(SEALEVELPRESSURE_HPA));
  Serial.println(" m");

  Serial.println();
  delay(2000);
}`

dherrada, 2020-06-23:
@Idahowalker Well, it definitely makes me feel better that this was an issue before 1.1.0, but it's still not good that's happening. I don't know C++, but I'll try to have someone who does help you with this.

Idahowalker, 2020-06-23:
As a note, I tried using https://github.com/SV-Zanshin/BME680/blob/master/examples/SPIDemo/SPIDemo.ino, that library works.

dherrada, 2020-06-24:
@Idahowalker You mean that the example you linked works with SV-Zanshin's library, right?

Also, I don't think you're meaning to close the issue every time. If you press the green 'comment' button instead of the 'close and comment' button, it won't close the issue

dherrada, 2020-06-24:
Also, if you don't mind, I've got one more thing I'd like you to try. Do you think you could give it a go with 1.0.7? It's possible that the regression happened in 1.0.8 so I'd like to see if that might fix it

Idahowalker, 2020-06-24:
> Also, if you don't mind, I've got one more thing I'd like you to try. Do you think you could give it a go with 1.0.7? It's possible that the regression happened in 1.0.8 so I'd like to see if that might fix it

I am back to using 1.0.7, which works.

I am not sure why 1.0.8 and above update the oled display driver and mess with core0.

dherrada, 2020-06-24:
Awesome. That's actually great news since we now know the regression happened in-between 1.0.7 and 1.0.8 which significantly narrows our search for the issue.

Thanks

dherrada, 2020-06-24:
Here are the changes from 1.0.7 to 1.0.8. It's a lot, but I'll try to take a look through and see if I can find anything that may have caused the issue.

https://github.com/adafruit/Adafruit_BME680/compare/1.0.7...1.0.8

sblantipodi, 2020-06-24:
I chime in.
If you see the difference between the 1.0.7 and 1.1.0 code
constructor changed without any breaking changes.

constructor changed from 
```
/**************************************************************************/
Adafruit_BME680::Adafruit_BME680(int8_t cspin)
  : _cs(cspin)
  , _meas_end(0)
{
  _BME680_SoftwareSPI_MOSI = -1;
  _BME680_SoftwareSPI_MISO = -1;
  _BME680_SoftwareSPI_SCK = -1;
  _filterEnabled = _tempEnabled = _humEnabled = _presEnabled = _gasEnabled = false;
}
```

to this:
```
/*!
 *  @brief  Instantiates sensor with i2c.
 *  @param  *theWire
 *          optional Wire object
 */
Adafruit_BME680::Adafruit_BME680(TwoWire *theWire)
    : _cs(-1), _meas_start(0), _meas_period(0) {
  _wire = theWire;
  _BME680_SoftwareSPI_MOSI = -1;
  _BME680_SoftwareSPI_MISO = -1;
  _BME680_SoftwareSPI_SCK = -1;
  _filterEnabled = _tempEnabled = _humEnabled = _presEnabled = _gasEnabled =
      false;
}
```

this can be the problem here imho.

Idahowalker, 2020-06-24:
> @Idahowalker You mean that the example you linked works with SV-Zanshin's library, right?
> 
> Also, I don't think you're meaning to close the issue every time. If you press the green 'comment' button instead of the 'close and comment' button, it won't close the issue

Correct, the SV-Zanshin's library worked, I wanted to prove that the BME 680 I have is still working. Thanks for the tip on closing.

I am currently using the Adafruit version 1.0.7 BME680 library which is working.

dherrada, 2020-06-24:
That's good to know. I'll have someone more familiar with this library and C++ in general come take a look at this and hopefully we'll be able to get the next version fully functional

siddacious, 2020-06-26:
@Idahowalker @dherrada after some unrelated issues I was able to reproduce this. I'll dig in and see what's up.

Idahowalker, 2020-06-26:
> @Idahowalker @dherrada after some unrelated issues I was able to reproduce this. I'll dig in and see what's up.

Thanks for looking into the issue.

ladyada, 2020-06-28:
@Idahowalker hi please try the latest commit here (you'll have to uninstall and then manually install this library)
https://learn.adafruit.com/adafruit-all-about-arduino-libraries-install-use

ladyada, 2020-06-29:
this library 
https://github.com/adafruit/Adafruit_BME680/archive/master.zip

Idahowalker, 2020-06-29:
I installed the new library from the zip. I uploaded the code to a ESP32 that does not have a BME680 attached. The new library did not fail on Adafruit_BME680 bme( GPIO_NUM_5 );. Yea.

Right now and for the rest of the day, its raining outside. The BME680 is physically attached to a weather station that's outside. I do not want to, at this time, open the weather station and load the new library to the weather station. When the rain stops, supposed to be tonight, I will bring the weather station inside, upload the new library, and test.  The good news is that the library did not, at instantiation, crash the ESP32. So bear with me, for another day-ish, for when the rain stops, to get the results of a full operational test.

ladyada, 2020-06-29:
ok!

Idahowalker, 2020-06-30:
I have uploaded the test library, which shows up as 1.1.0. I did not experience any instantiation errors and is producing a result set. I will let the device run, inside, for several hours, then move the device outside if I do not get any BME680 library issues. I really do not expect for their to be issues at this point.

Idahowalker, 2020-06-30:
I have closed and sealed the weather station electronics and place the weather station outside to assume normal operations. I do no foresee any more issues but if I do.

Oh, Thanks.

ladyada, 2020-06-30:
great!

closed 2020-06-30 by ladyada