adafruit/Adafruit_BME680 #50: Altitude formula seems to be incorrect

cowboydaniel, 2022-08-01:
After doing a heap of research and loads of math, it turns out that it looks like this library for the BME680 does an incorrect equation for altitude. As far as I can work out, it should be more like this:
`float Alt = ((((((10 * log10((bme.pressure / 100.0) / 1013.25)) / 5.2558797) - 1) / (-6.8755856 * pow(10, -6))) / 1000) * 0.30);`

This returns the altitude in meters above sea level, and should be accurate to 39 kilometers (high enough for a weather balloon).

I got the formula from [here](https://www.brisbanehotairballooning.com.au/pressure-and-altitude-conversion/).

I'm not sure who to contact for this library, so I'm posting it here.

Originally, I was trying to figure out why this sensor was saying that I live 21 meters below the sea level instead of 44 meters above sea level, I was quite surprised when I pulled apart the formula and found out it is actually for a completely different sensor!

cowboydaniel, 2022-08-01:
edit: 
This code works better:
`if (! bme.performReading()) {
    Serial.println("Failed to perform reading");
    return;
  }
  
  float P1 = (bme.pressure / 100.0);
  float h = 44;
  float T = bme.temperature;
  float P0 = (P1 * pow(1 - ((0.0065 * h) / ((T + (0.0065 * h)) + 273.15)), -5.257));

  Serial.print("Current pressure: ");
  Serial.println(P1);
  Serial.print("Current sea level pressure: ");
  Serial.println(P0);
  Serial.print("Current temperature: ");
  Serial.println(T);
  float P = bme.pressure;
  float Alt = ((((((10 * log10((P / 100.0) / P0)) / 5.2558797) - 1) / (-6.8755856 * pow(10, -6))) / 1000) * 0.30);
  Serial.print("Altitude Formula: ");
  Serial.println(Alt);`

caternuson, 2022-08-01:
See here for some discussion on computing altitude from pressure:
https://learn.adafruit.com/clue-altimeter/computing-altitude

The current equation in general looks OK. The coefficients are not exactly the same, but should only cause minor differences. Could try changing to:
```cpp
  return 44307.69396 * (1.0 - pow(atmospheric / seaLevel, 0.190284));
```
and see if it helps.

But the main thing to pay attention to is properly setting the `seaLevel` value. The same guide linked above goes into lengthy discussion about this. Did you determine your current and local value for `seaLevel`?

caternuson, 2022-08-01:
Cross posted:
https://forums.adafruit.com/viewtopic.php?f=19&t=193260

cowboydaniel, 2022-08-02:
> See here for some discussion on computing altitude from pressure: https://learn.adafruit.com/clue-altimeter/computing-altitude
> 
> The current equation in general looks OK. The coefficients are not exactly the same, but should only cause minor differences. Could try changing to:
> 
> ```c++
>   return 44307.69396 * (1.0 - pow(atmospheric / seaLevel, 0.190284));
> ```
> 
> and see if it helps.
> 
> But the main thing to pay attention to is properly setting the `seaLevel` value. The same guide linked above goes into lengthy discussion about this. Did you determine your current and local value for `seaLevel`?

Yes, that is what the first of the two formulas does. It determines the current sea level pressure by using the current altitude pressure. It then uses that to work out the altitude in meters above sea level. 

I wasn't sure which forum to post it on, so I put it on both. Is that an issue @caternuson ?

ladyada, 2022-08-02:
?? you cannot calculate sea level pressure from current altitude pressure and then use the same data to calculate altitude. you need to have one or the other

cowboydaniel, 2022-08-02:
You need to actually run the code I posted above, as it works really well for me. I do not use the "current altitude" pressure as such, it uses the current pressure **no matter** the altitude I am at. No where do I use the altitude in my formulas as I do not need to.

caternuson, 2022-08-02:
How did you determine your current and local value for sea level pressure when using this library to compute altitude?

Specifically, what value did you supply as the parameter to `readAltitude()` and where did it come from?
```cpp
  float readAltitude(float seaLevel);
```

cowboydaniel, 2022-08-02:
@caternuson 
`float P1 = (bme.pressure / 100.0);
float h = 44;
float T = bme.temperature;
float P0 = (P1 * pow(1 - ((0.0065 * h) / ((T + (0.0065 * h)) + 273.15)), -5.257));`

P0 replaces seaLevel. However I haven't put it in the library, I've replaced it only in code ie. `bme.readAltitude(P0);`

This makes it change without me having to edit anything, but it only works if my altitude doesn't actually change. 

I am only suggesting that the formula be more accurate in the library. I'm not ordering it. I also realize that I could be wrong in how I am doing it.

If no-one likes that, I will replace it in my library and never mention the subject again. So far I've been getting a lot of backlash about this and I am getting a bit fed up. I am only trying to get this sensor more accurate without changing a line of code every time my local areas sea level pressure is updated (every hour in my case).

This line here really annoys me `float h = 44;` as it can't be a constant, but I don't know how to make it a variable without incorporating a GPS module into this sensor.