adafruit/Adafruit_BME680 #52: Re#defined items in BSEC_Software_Library/src/bme680/bme680_defs.h generate warnings

ayourk, 2022-11-21:
https://github.com/adafruit/Adafruit_BME680/blob/29897a675e3102ae8e5a329e58377a4efdb36c3b/Adafruit_BME680.h#L38-L60

The above lines are defined in BSEC_Software_Library/src/bme680/bme680_defs.h and cause warnings if both `bsec.h` is included and  Adafruit_BME680.h is included in the same project.

caternuson, 2022-11-21:
Not familiar with BSEC, but appears to be a more sophisticated library for the BME680 provided by Bosch:
https://www.bosch-sensortec.com/software-tools/software/bsec/
https://github.com/BoschSensortec/BSEC-Arduino-library

The BSEC library is probably meant to be used stand alone. Looks like there's a basic example here:
https://github.com/BoschSensortec/BSEC-Arduino-library/blob/master/examples/basic/basic.ino

caternuson, 2022-11-23:
Closing. Looking at this again, pretty sure the BSEC library is meant to be standalone. That is - don't use the BSEC library in conjunction with this Adafruit Arduino library.

closed 2022-11-23 by caternuson