adafruit/Adafruit_BME680 #53: Possible humidity issue on Espressif 4.4, 5.3 and 6.0 framework

hpsaturn, 2023-01-28:
## Overview

Over Espressif framework 3.2.0 this library works fine. But using the versions above, the humidity only set the value on 100%.

## Details

- Implementation:  like a bme680async example
- Tested on PlatformIO

### Espressif 3.2.0

**Dependencies and libraries**:

```
Adafruit BusIO @ 1.14.1 (required: Adafruit BusIO)
Adafruit Unified Sensor @ 1.1.7 (required: Adafruit Unified Sensor)
Adafruit BME680 Library @ 2.0.2 (required: adafruit/Adafruit BME680 Library @ 2.0.2)
Platform espressif32 @ 3.2.0 (required: espressif32 @ 3.2.0)framework-arduinoespressif32 @ 3.10006.210326 (required: platformio/framework-arduinoespressif32 @ ~3.10006.0)
framework-espidf @ 3.40200.210118 (required: platformio/framework-espidf @ ~3.40200.0)
```

### Espressif 4.4.0

**Dependencies and libraries**:

```
Platform espressif32 @ 4.4.0 (required: espressif32 @ 4.4.0)
framework-arduinoespressif32 @ 3.20003.220626 (required: platformio/framework-arduinoespressif32 @ ~3.20003.0)
```

** CanAirIO Sensors Library output:**

```bash
13:57:43.243 > -->[SLIB] attempt enable sensor  : BME680
13:57:43.243 > [  2788][W][Wire.cpp:204] begin(): Bus already started in Master Mode.
13:57:43.581 > -->[SLIB] sensor registered      : BME680        :D

13:58:00.327 > -->[SLIB] Preview sensors values : PM1:1.0 PM2.5:1.0 PM10:1.0 T:27.6 H:100.0 P:785.0 Gas:0.0 Alt:2102.1
``` 

** Implementations: **

Setup: https://github.com/kike-canaries/canairio_sensorlib/blob/esp32c3test/src/Sensors.cpp#L1417-L1426
Read: https://github.com/kike-canaries/canairio_sensorlib/blob/esp32c3test/src/Sensors.cpp#L874-L891

caternuson, 2023-01-29:
Can you reproduce this using the Arduino IDE (and not Platform IO)?

hpsaturn, 2023-01-29:
I'm not using Arduino IDE, but on the other hand both tools using the same Espressif framework, the same toolchain, the same versions.. I don't understand the point.  Can you give me the version number that you have in Arduino IDE please?

caternuson, 2023-01-29:
This is an Arduino library, so need to demonstrate the issue with Arduino. Any recent version of the IDE is fine, like 1.8.19 or latest 2.x release.

Also the simplest Arduino sketch that can be used to demonstrate the issue. Can just be the example from the library:
https://github.com/adafruit/Adafruit_BME680/blob/master/examples/bme680test/bme680test.ino
if that shows the issue.

rodri16, 2024-06-11:
I reproduced the error, after a while I get reading error using the simplest example on Arduino IDE 2.3.2 and esp32 by espressif 2.0.17.  I am using Heltec Lora32 v3 ESP32-s3 based board

After "Failed to perform reading :(" bme values are fixed at a wrong value,
Temperature = 33.15 *C
Pressure = 773.86 hPa
Humidity = 100.00 %
Gas = 0.00 KOhms

This is the logging

> BME680 test
Init Result: 0
T1 = 26243
T2 = 26635
T3 = 3
P1 = 35770
P2 = -10370
P3 = 88
P4 = 5392
P5 = 73
P6 = 30
P7 = 14
P8 = 733
P9 = -3981
P10 = 30
H1 = 792
H2 = 1007
H3 = 0
H4 = 45
H5 = 20
H6 = 120
H7 = -100
G1 = -20
G2 = -19363
G3 = 18
G1 = -20
G2 = -19363
G3 = 18
Heat Range = 1
Heat Val = 44
SW Error = -2
SetConf Result: 0
SetConf Result: 0
SetConf Result: 0
SetConf Result: 0
SetConf Result: 0
SetHeaterConf Result: 0
Opmode Result: 0
SetConf Result: 0
SetConf Result: 0
SetConf Result: 0
SetConf Result: 0
SetHeaterConf Result: 0
Opmode Result: 0
Waiting (ms) 182
t_fine = 0.00
Getting sensor data
GetData Result: 0
data.status 0xB0
Temperature = 22.25 *C
Pressure = 979.23 hPa
Humidity = 68.86 %
Gas = 246.21 KOhms
Approx. Altitude = Opmode Result: 0
Waiting (ms) 182
t_fine = 113939.15
Getting sensor data
GetData Result: 0
data.status 0xB0
287.08 m

> Opmode` Result: 0
Waiting (ms) 182
t_fine = 104714.58
Getting sensor data
GetData Result: 0
data.status 0xB0
Temperature = 20.34 *C
Pressure = 979.33 hPa
Humidity = 68.33 %
Gas = 198.41 KOhms
Approx. Altitude = Opmode Result: 0
Waiting (ms) 182
t_fine = 104129.21
Getting sensor data
GetData Result: 0
data.status 0xB0
286.23 m

> Opmode Result: 0
Waiting (ms) 182
t_fine = 104725.96
Getting sensor data
GetData Result: 2
Failed to perform reading :(
Opmode Result: 0
Waiting (ms) 182
t_fine = 104725.96
Getting sensor data
GetData Result: 0
data.status 0x80
Temperature = 33.15 *C
Pressure = 773.86 hPa
Humidity = 100.00 %
Gas = 0.00 KOhms
Approx. Altitude = Opmode Result: 0
Waiting (ms) 182
t_fine = 169750.55
Getting sensor data
GetData Result: 0
data.status 0x80
2216.41 m

hpsaturn, 2024-06-13:
thanks for your feedback.. I notice that the issue on some sensors happens.. but also in some not.. Also some users reported that the sensor has problems with environments with to much humidity.. On the other hand with old esp32 we have better results and the firmware is more stable, when you use the Espressif framework 4.4.0.

rodriogs, 2024-06-14:
Is it a hardware problem or a library problem?

hpsaturn, 2024-06-14:
In the moment that I published this ticket was very clear that was a Library problem because I remember that the sensor was in indoors and only changing the Toolchains or the Espressif framework you was able to replicate it. Now, one year and half after, I don't know.. I notice that my indoor sensor is working well with 4.4.0. Maybe we could close this issue, but on the other hand seems that some users have the same problem, and also I have a remote device that have the same behavior.. Maybe we have two different issues?

caternuson, 2024-06-18:
Closing since it sounds like original issue is resolved.

@rodri16 Your behavior is something different and seems related to a failed reading attempt. Please open a new issue.

closed 2024-06-18 by caternuson